pottedmeat7/OpenVR-WalkInPlace #98: How to disable controller based movement?

morinehtar0, 2025-09-27:
Hi,
I'd like to do the walking with only my two trackers on my ankles, but I can not create a model without at least one controller.
when trying to walk in-game I have to move that controller as well as my feet.
But I don't want my walking to be dependent on the controller; is there a way to disable/remove the controller?

pottedmeat7, 2025-09-28:
I believe there should be a way to set or select your tracker as  the controller. I think you have to set the tracker as a "tracked hand". Even though it may be on your ankle.

morinehtar0, 2025-09-28:
Thanks for the reply, but could you maybe be a bit more specific? Where would I set the tracker as "tracked hand"?

And why does there have to be a controller in the model? couldn't a model be made with just the HMD and some trackers?

pottedmeat7, 2025-09-28:
Ah yes forgot looks like the wording did change at some point. It would be the bottom row on this screen (although the screenshot is a bit out of date)
https://github.com/pottedmeat7/OpenVR-WalkInPlace/blob/master/docs/screenshots/full_setup/ovrwip_step2.png

If you select the right combination of things on this screen it should allow your trackers to replace the functionality of your controllers. I think you should select items on the bottom row, and de-select the controllers on the top row.

If you make a selection here and save it as a profile the device ids would be in the text profile in your AppData directory.

In that text profile there should be two fields `cntrl1Idx` and `cntrl1Type`  

Those values should have the values of the tracker you choose to replace your controller.

morinehtar0, 2025-10-02:
> Now could I just set cntrl1Idx to -1 ?

Tested it; nope, doesn't work.

morinehtar0, 2025-10-04:
Ok, thanks. I got it to work somewhat. I can now walk around, only I would need to do some tweaking because starting and stopping now takes a couple of seconds. (noticed some settings in the docs, so should be allright I guess)
One other thing, though: I am walking in the direction I'm looking instead of the direction I'm walking, which is not great.

But now the dev from NALO is back and fixed it's problem, so I'm sorry, but going back to that...

morinehtar0, 2025-10-05:
The latter; I want to be walking with only my feet, leaving my hands (and head) free to do other things.